mROS-base/mros2-esp32 #6: mros-esp32 introspection

LuisYordano, 2023-06-21:
Hi team mros2-esp32.

I am testing the demo: echoback_string, which runs correctly.

-------------------
[image]

--------------------

I report the following:

1-Cannot see in the console the node that is currently running: mros2_node.

2-Node and topics cannot be visualized in rqt: /to_linux and /to_stm

3-The data cannot be displayed in the topic monitor.

-------------
[image]

--------------
[image]

--------------

It is correct that this information cannot be displayed, or it is necessary to do some additional configuration?

I thank you in advance for your answer.

takasehideki, 2023-09-07:
@LuisYordano thank you so much for trying our mros2 world! However, I'm truly sorry for replying so late,,, ;(

1. For node names, we, unfortunately, observe them due to the limitation of embeddedRTPS (RTPS stack used for mROS 2).  Please check the Issue in `mros2` repository (I just created it). https://github.com/mROS-base/mros2/issues/44
2. When you want to operate `echoback_string` on the mros2 device, you need to operate `mros2_echoreply_string` on the ROS 2 host machine. Sorry there is no description in this repository, but you can find the operation as follows (we plan to enrich documentation, or commonize the sample code).
  - https://github.com/mROS-base/mros2-mbed#echoback_string-default
  - https://github.com/mROS-base/mros2-host-examples#mros2_echoreply_string
3. As far as our experience, you can observe the message of topics by the topic monitor. Please operate along with 2.

LuisYordano, 2023-09-19:
Ok, @takasehideki.

closed 2023-09-19 by LuisYordano