mROS-base/mros2-esp32 #8: It would be nice to see the ros2 node list. (mros2-esp32)

yamati-kz, 2023-07-14:
Only two microcomputers are connected. They are seeed studio xiao esp32c3 and ESP32-S3-DevKitC-1.
There seems to be no problem with pub/sub between two microcomputers.

It would be nice to see the ros2 node list. (mros2-esp32).

[image]

Here is the source code outline.
[image]

takasehideki, 2023-09-07:
@yamati-kz thank you so much for trying our mros2 world! However, I'm truly sorry for replying so late,,, ;(

As you pointed out, it would definitely be great if we could check the mros2 node names by `ros2 node list`. However, as far as we observed, this is not going to be easy to achieve,,, Since this is a limitation of the mROS 2 library, I created Issue in that repository. https://github.com/mROS-base/mros2/issues/44

You may leave this Issue open, as it is a good guide for those who will be trying mros2-esp32.

yamati-kz, 2024-04-02:
This issue will be closed. I look forward to the future.
Since it is not resolved, refer to this https://github.com/mROS-base/mros2/issues/44.

closed 2024-04-02 by yamati-kz