mROS-base/mros2-esp32 #19: CMakeLists Command Escapes Python Virtual Environment

Foxei, 2023-12-13:
# Short Summary
The CMakeLists file escapes the Espressif Python virtual environment when generating the template.hpp, if this project is used as a component.

## Current State

The provided code generates the template.hpp when using mros2-esp32 as a component. It executes within the host's Python3 environment, and not the Espressif IDF virtual Python environment set by the export.sh script before the build step.
```cmake
add_custom_command(
        OUTPUT templates.hpp
        COMMAND python3 ${COMPONENT_PATH}/mros2/mros2_header_generator/templates_generator.py --outdir ${COMPONENT_PATH}/platform --indir ${CMAKE_SOURCE_DIR}/main)
add_custom_target(Template SOURCES templates.hpp)
add_dependencies(${COMPONENT_LIB} Template)
```

## Proposed Change

I suggest forcing CMake to use the Espressif IDF Python virtual environment to avoid mixing the host and the virtual environment. The following change makes this possible.
```cmake
set(PYTHON "$ENV{HOME}/.espressif/python_env/idf5.2_py3.9_env/bin/python")
add_custom_command(
        OUTPUT templates.hpp
        COMMAND ${PYTHON} ${COMPONENT_PATH}/mros2/mros2_header_generator/templates_generator.py --outdir ${COMPONENT_PATH}/platform --indir ${CMAKE_SOURCE_DIR}/main)
add_custom_target(Template SOURCES templates.hpp)
add_dependencies(${COMPONENT_LIB} Template)
```

# Final Remarks

In order to provide clarity on the expected behavior and environment requirements, the documentation should be updated to include information about the necessary Python environment. Currently, it only mentions the need to install Jinja2 not in which environment.

I'm more than happy to assist and create the pull request if this behavior aligns with the project's intentions. Please let me know your thoughts, and I'll proceed accordingly. ✌🏼

Simon

Foxei, 2023-12-13:
Obviously I will not hardcode the `idf5.2_py3.9_env` version numbers in the pull request.

takasehideki, 2024-12-29:
@Foxei thank you for your proposal!! 
One of my concerns is that the version number should be hardcoded. The version of ESP-IDF that this repository assumes is [5.1.1](https://github.com/mROS-base/mros2-esp32/blob/4d7eca75d65d440b427d60048990d189201c384b/README.md?plain=1#L38) (which is old and should be updated,,,) and the version that each user usually uses may differ.
We welcome your idea as a PR that can address this concern!