mROS-base/mros2-esp32 #24: Successfully pub/sub with mros2-esp32 using Seeed Studio XIAO ESP32S3

yamati-kz, 2024-08-19:
I tried mros2-esp32 on Seeed Studio XIAO ESP32S3, so I'll report on it. This is not an Issues report.
Verification requested by Seeed Studio.
[image]

Seeed Studio XIAO ESP32S3. ( ESP-IDF v5.4-dev-1832-g23e4823f17 )
NVIDIA Jetson Orin Nano Developer Kit. ( NVIDIA Jetpack 6.0 first production release / Ubuntu 22.04.4 LTS )

mros2-esp32 : Latest version as of August 2, 2024
[mros2-host-examples](https://github.com/mROS-base/mros2-host-examples) : Latest version as of August 2, 2024
[ROS 2 Humble Hawksbill](https://docs.ros.org/en/humble/index.html)

1. The communication log is attached.
[image]

[image]

2. Commands used in Seeed Studio XIAO ESP32S3
```
. $HOME/esp/esp-idf/export.sh
idf.py set-target esp32s3
idf.py menuconfig
idf.py build
sudo chmod 666 /dev/ttyACM0
idf.py -p /dev/ttyACM0 flash
idf.py -p /dev/ttyACM0 monitor
```
3. Seeed Studio XIAO ESP32S3 menuconfig settings
[image]
[image]

yamati-kz, 2024-08-19:
[image]

meilily, 2024-08-21:
This is superb!! Could you also include XIAO ESP32S3 and XIAO ESP32C6 to [the supported boards](https://github.com/mROS-base/mros2-esp32) in the readme of this repo? Thank you so much!!!!

yamati-kz, 2024-08-21:
@meilily : Thank you for your message here. Unfortunately, I am not a contributor and cannot rewrite the Readme.

@takasehideki , @s-hosoai : Takase-san and Hosoai-san, I would appreciate it if you could consider this matter.

yamati-kz, 2024-08-22:
https://x.com/seeedstudio/status/1826175627063603496?s=46&t=4HGyVKsDUAlfHEv_mPnMUw

closed 2024-09-28 by yamati-kz

yamati-kz, 2024-12-29:
@meilily 
The mros2-esp32 project has been listed in the top README!

meilily, 2024-12-31:
Thank you so much!